pablozg/freeds #113: Bug: "Failed to read term temperature from DS18B20 sensor" message and automatic control interruption

Csontikka, 2024-10-28:
Bug: "Failed to read term temperature from DS18B20 sensor" message and automatic control interruption
Version: 1.1.0020 Beta Build: 05042024 Web: 1.1.5
Hardware: PWM10A 2200W MOSFET Trailing Edge Dimmer + ESP32 with additional cooling (relay4)
Connected Temperature Sensor: DS18B20 (for measuring boiler temperature)

Description of the Issue:
On 2024-10-28 at 16:33, the error message "Failed to read term temperature from DS18B20 sensor" appeared in the console (see attached image), which led to the interruption of automatic PWM control. PWM was enabled in AUTO mode. Despite this error, I continued to receive real-time boiler temperature data in Home Assistant, suggesting that the issue was transient; however, it still resulted in a control shutdown.

Approximately 2 hours later, at 18:49, I noticed the issue and saved the configuration through the device’s web interface, which I believe restarted the internal processes. Following this, the device resumed operation immediately and has been running without issues since.

Interesting Observation:
While the DS18B20 error interrupted the automatic control, temperature data continued to update in Home Assistant. I suspect the error was temporary, yet it still blocked the system's operation. The device is equipped with additional cooling (relay4), so overheating could not have caused this issue.

Attached Images:
The attached images display the error messages in the console and the data from the two relevant sensors (boiler temperature and PWM percentage values).

[image]
[image]

Request:
I am looking for a solution that prevents a momentary sensor error from stopping the entire system. If a sensor occasionally gives a faulty reading, it should not lead to a complete control shutdown.

If you need any further information, please let me know!

Regards,
Csontikka

Csontikka, 2024-11-06:
The issue has recurred 9 days later. The device exhibited the exact same behavior as before, so I won't go into detail about the events. However, this suggests that the problem is not isolated but rather a recurring, systematic issue that requires further investigation.

Thank you in advance for taking the time to address this!

pablozg, 2024-11-06:
Hola, no especificas la configuración que tienes en el apartado de temperaturas para tratar de replicarlo, aunque en mi opinión no es un fallo como crees, lo más probable (al no disponer de tu configuración) es que la temperatura del termo estuviera dentro del rango configurado (por encima de los 50 grados en tu gráfica) por lo que una vez detenido por el fallo de lectura del termo, este no se pone a funcionar hasta que la temperatura esté por debajo del umbral de activación (vuelvo a repetir que no has puesto una captura de tu configuración), por lo que una vez hubiese llegado a esta temperatura se habría activado de nuevo hasta llegar a la temperatura de corte.

Saludos.

Csontikka, 2024-11-07:
Saludos,

I have taken multiple screenshots of the settings that seem relevant, which I have not modified since. The boiler's heating range was set between 30 and 70 degrees, and the temperature never approached these values. However, the previously recorded boiler temperature graph clearly shows that the control did not stop when reaching a specific temperature. The temperature remained unchanged when the PWM turned off.

Regards,
Csontikka

[image]
[image]
[image]
[image]

Csontikka, 2024-11-07:
Dear Pablozg,

This issue continues to concern me, so I took a detailed look at the error logs from November 6. I noticed one additional detail that might provide a useful clue for you.

The graphs clearly show that at the moment the error occurred, the status of the "FreeDS ojle Error" sensor briefly changed to "Problem." 

Additionally, just a few minutes before the error, at the same temperature, the PWM was still switching correctly.

I've also taken a screenshot related to the specific time of the issue, which shows the status of the sensors. I hope this visual evidence, along with the details provided, can assist in identifying the problem.

[image]

I tried to annotate the screenshot as thoroughly as possible to highlight the key points.

Regards,
Csontikka

pablozg, 2024-11-09:
Hola,

Con la gráfica me estás dando la confirmación de lo que te indiqué en un principio, la temperatura antes de dar el error de sonda era de 44.5ºC, tu temperatura de activación es inferior a 30ºC.

Cuando da el error de temperatura el sistema se detiene, y una vez obtiene la temperatura de nuevo, como la temperatura actual del termo no es inferior a 30ºC no se activa de nuevo el gestor (esta es la programación establecida).

Si la temperatura del termo llegará a menos de 30ºC entonces volvería a activar el gestor hasta llegar a los 70ºC que tienes configurado.

Cuando reinicias el gestor o cambias la configuración se prioriza el entregar potencia hasta que se vuelvan a cumplir las condiciones.

En todo caso, trataré de pensar una forma para que en caso de dar error la sonda de temperatura, permita seguir calentando si no ha llegado a la temperatura de corte.

Saludos.

Csontikka, 2024-11-09:
Dear Pablozg,

Now I clearly understand the issue. The control stops due to the sensor error, and one of the conditions for it to restart (the "turn on temperature") is not met.

So, my workaround is simple: I will lower the "turn on temperature" value to 10°C, which will always be fulfilled. And of course, I’d be happy if a permanent solution can also be found.

[image]

Regards,
Csontikka

pablozg, 2024-11-10:
Hola, no has entendido correctamente como funciona el gestor, si colocas la temperatura de encendido a 10ºC no funcionará el gestor hasta que la temperatura del agua esté por debajo de 10ºC, debes aumentar la temperatura a por ejemplo a 45ºC, entonces cuando la temperatura del agua esté a 44.5ºC o menos entonces empezará a calentar hasta llegar a los 70ºC.

Otra opción es activar la opción de temperatura mínima, de forma que siempre tendrás el agua a esa temperatura (enciende a la temperatura conectada y para cuando la supera en 3ºC).

Saludos.

Csontikka, 2024-11-10:
Dear pablozg,

You are absolutely right. After I wrote my comment yesterday, I realized in the evening that what I wrote was nonsense. :)

I will try your suggestion. Thank you.

Regards,
Csontikka

pablozg, 2024-11-11:
Hola, por favor contacta conmigo mediante telegram y te pasaré una versión de prueba.
Saludos.

ruisilva93, 2025-04-05:
I’m having the same problem: it’s bellow the max temperature, the sensor gives an error and PWM stopes until a restart or about 2h. But I still that temperature visível through that sensor on freeds. But 0% power while have exceeding powers available. It’s clearly a bug. Might be hard to reproduce, but it’s happening for more than one. 

I’m version .20 too.

closed 2025-11-23 by pablozg