pablozg/freeds #116: Feature Request: Configurable Minimum Run Time for Relay Outputs

Csontikka, 2024-11-06:
**Software:** FreeDS v1.1.0020 Beta Build: 05042024 Web: 1.1.5
**Hardware:** PWM10A 2200W MOSFET Trailing Edge Dimmer + ESP32 with additional cooling (relay4)
Connected Temperature Sensor: DS18B20 (for measuring boiler temperature)

**Description:** Currently, in my setup, I use the 4th relay output to control an external fan, which is configured to turn on when the PWM level exceeds 30%. While the system works as intended, I have noticed that frequent fluctuations around this threshold cause the fan to rapidly switch on and off. This behavior could shorten the fan's lifespan and even lead to operational issues for more significant loads.

Proposed Solution: I would like to request an additional setting in FreeDS to configure a minimum run time for each relay. This setting would ensure that once the relay output activates, the connected device (such as a fan) remains powered for a specified minimum duration before turning off, regardless of changes in the PWM level. Alternatively, another approach could be to implement a minimum switching interval, which would guarantee a minimum time between two consecutive state changes of the relay. Either option would mitigate issues caused by rapid toggling, providing operational stability and protecting connected devices from damage.

**Benefits:**

Protect Equipment: Extends the lifespan of connected devices by reducing wear from frequent on/off cycles.
Operational Stability: Ensures smoother operation by preventing excessive switching.
Versatile Configuration: Offers flexible settings per relay output to accommodate different equipment needs.
Example Use Case:

For a fan controlled by a relay, if the PWM level triggers the relay, the fan would run for a minimum of, say, 60 seconds, even if the PWM drops below the threshold. Alternatively, a minimum switching interval would ensure that the relay does not toggle again until a set time has elapsed.

Thank you for your efforts and contributions to the project and its development!
Regards,
Csontikka

pablozg, 2024-11-06:
Hola,

Las salidas ya poseen un retardo en el encendido y apagado que no se aplica a la salida 4 cuando está configurada como ventilador, lo añadiré a la lista TODO.

Saludos.

Csontikka, 2024-11-07:
Thank you for your positive response and for considering the suggestion! I'm glad to see it being added to the development list, and I’m confident it will enhance the system’s performance. I appreciate your prompt feedback!