pablozg/freeds #118: Beta 21 rompe integración con HASS

IvanSanchez, 2024-11-18:
Me pasan nota de que al actualizar a la beta 21 se rompe la integración con Home Assistant:  https://gitlab.com/IvanSanchez/homeassistant-freeds/-/issues/32

¿Qué ha cambiado en la API de Websockets de la beta20 a la beta21?

pablozg, 2024-11-21:
Si, ya no uso websockets para obtener la información, ahora se hace un GET al endpoint "json":

Devolviendo:

`{"Web":{"Oled":1,"screenBrightness":254,"POn":1,"PwmMan":0,"SenTemp":0,"Msg":0,"pwmfrec":30000,"pwm":7,"loadCalcWatts":"24","baudiosMeter":9600,"workingMode":44,"workingModeName":"IBMT","masterMode":44,"masterModeName":"IBMT","invertedSign":false,"tempShutdown":0,"error":0},"Relays":{"R01":true,"R02":false,"R03":false,"R04":false},"Inverter":{"wsolar":"2410.38","wgrid":"95.55","wtoday":"1.74","gridv":"240.57"},"Meter":{"mvoltage":"240.57","mcurrent":"1.35","mpowerFactor":"0.29","mfrequency":"50.00"},"Energy":{"KwToday":"0.000","KwYesterday":"9.017","KwTotal":"52.432","KwExportToday":"0.030","KwExportYesterday":"9.556","KwExportTotal":"57.226"},"Temperature":{"tempTermo":"-127.0","tempTriac":"-127.0","tempCustom":"-127.0","customSensor":"Temp. Ambiente"}}`

IvanSanchez, 2024-11-21:
Pues un cambio así *al menos* debería estar en el changelog.

pablozg, 2024-11-21:
Esa es tu opinión, pero no pienso entrar en polémicas, has preguntado y te he contestado.

closed 2024-11-21 by pablozg