pablozg/freeds #129: Battery or grid solar surplus

ruisilva93, 2025-04-05:
I have a battery system. I have version .20 but it’s the same in .21 (I have tried)
I have set it up for when battery it’s above 30% and solar it’s above 500w (potência objetivo).

So, if solar it’s 2000w we have 500w to battery, and 1500w to boiler. Good! Perfect! Awesome!!!

But when the battery it’s 100%, and can not receive 500w anymore, the PWM sends 0w to boiler, even with solar over 4000w and 4000w going to the grid.

If I change the settings to not consider battery, it will first charge the battery 100% and only then start to heat the boiler (witch in rainy days it’s not ideal once the water will take longer to heat, or battery might not reach 100% at all).

Can you enable an OR condition, were the two conditions that you already have work together? And even when the battery control it’s turned on, it understands and uses grid power?

pablozg, 2025-04-07:
Hola, en la próxima versión que publicaré (2.0.0), incluyen muchos cambios importantes, como el soporte para homeassistant mediante mqtt de forma nativa, nuevos inversores y la solución a tu problema.

Cuando la batería llegué al 100% no dejará de derivar hasta que no baje del 100% o supere los 100W de consumo de batería (consumo máximo programado mientras esté al 100% la batería), cuando el % baje del 100% se aplicarán los criterios configurados en la web.

Saludos.

ruisilva93, 2025-09-04:
Any update on this?